tomasz-lewicki/ai-thermometer #2: where is requirements.txt

jazz-ai, 2020-08-12:
need requirements.txt

tomek-l, 2020-08-12:
Hi @jazz-ai !

The ```pip3 install -r requirements.txt``` was leftover from the previous AI Thermometer version that depended on ROS, as well as some other libraries. I will delete that line from the instructions.

Current version has no external requirements other than OpenCV 4.4  and nvidia Jetpack 4.4.

Let me know if you have any other issues.

jazz-ai, 2020-08-14:
I tried to run main.py
It is showing 
**uvc_open error -3**
and unable to start anything

tomek-l, 2020-08-16:
It means that your user doesn't have r/w access to the USB ports, so it cannot access the Pure Thermal device.

junartho, 2020-08-26:
hi @tomek-l 
nice codes. please tell me how to handle r/w access on usb ports.

tomek-l, 2020-08-26:
Hi @junartho 

If you want a quick and dirty solution do:
```shell
echo 'SUBSYSTEM=="usb",  ENV{DEVTYPE}=="usb_device", GROUP="plugdev", MODE="0664"' | sudo tee /etc/udev/rules.d/10-libuvc.rules 
sudo udevadm trigger
```
(this gives camera access to all users)

For a better, fine-grained permission setup, create a new group, give permissions to that group only and your user to the newly created group. You can read more [here](http://wiki.ros.org/libuvc_camera#Permissions).

tomek-l, 2020-08-26:
Let me know if that solved the problem :slightly_smiling_face:

junartho, 2020-08-26:
thank you for very fast response. 👍

closed 2020-08-26 by tomek-l