tomasz-lewicki/ai-thermometer #3: Version 0.1: SSD (COCO) + FLIR Lepton 3.5 temp. measurement (~12FPS w/ GPU accel.)

junartho, 2020-08-27:
hi @tomek-l 

First of all, i wanna thank you for your codes. it really helps the communities at my place in this pandemic situation. however, these codes could only measure temperature with face straight to the camera and no mask used. i wonder if you have codes version 0.1 with SSD detection. I would appreciate it if you want to upload it.

tomek-l, 2020-08-28:
Hi @junartho ,

I didn't keep track of all the versions in parallel (v0.1, v0.2 and v0.3) but instead just overriding the repository with a newer version.  
You can view the v0.1 here: [https://github.com/tomek-l/ai-thermometer/tree/3bc78c2d7ce71a5f8fc7564c0c246c59196bae4e](https://github.com/tomek-l/ai-thermometer/tree/3bc78c2d7ce71a5f8fc7564c0c246c59196bae4e)

Keep in mind that the version 0.1:
1. Uses an SSD model, which is trained on the COCO dataset. While it gives you more classes (91 classes), it doesn't contain a "face" class, but rather a "person" class. This is less useful for the temperature estimation.

2. It uses ROS2 for communication. I dropped this approach in favour of communicating between threads. ROS2 is a bit of an overkill for this project, and folks might be unfamiliar with it. 

3. The setup is more involved, and the README is not as thorough.

I'm happy to help if you have any issues getting the v0.1 to work.

tomek-l, 2020-09-01:
Closing for now, but feel free to reopen if you have any further questions.

closed 2020-09-01 by tomek-l