tomasz-lewicki/ai-thermometer #4: I am implementing ai thermometer with your kind explanation.

kos8108, 2020-08-28:
I am implementing ai thermometer with your kind explanation. 

Now I am building the environment of Raspberry pi camera v2 & lepton2.5 & PureThermal2.

But nothing has happened since Uvc_open error -3 appeared after python3 main.py. 

Can I know the solution? 

Of course, the individual operation of each sensor was confirmed.

thank you.

tomek-l, 2020-08-28:
Hi,

Did you change the permissions for the USB camera? (There's a short
explanation in readme.md)

kos8108, 2020-08-28:
Thank you so much.

It was executed with sudo without any problems.

Additionally, the temperature value of the ir sensor cannot be output, which part should I check?

Thank you always for your answer.

tomek-l, 2020-08-28:
Hey,

So if you wish to read the temperature value, you will need to add
something like this to the main loop:

print(ir_thread.frame.mean())

This will print the mean temperature of the 120x160 pixel array.

If you want to access the temperature of individual pixels, you'd do:

print(ir_thread.frame[0][0])

This way you'll access the value of the (0,0) pixel in degrees Celsius.

Hope this helps!

tomek-l, 2020-09-01:
Closing for now, but feel free to reopen if you have any further questions!

closed 2020-09-01 by tomek-l