tomasz-lewicki/ai-thermometer #5: Couldn't get correct temperature

weshao, 2020-08-31:
Hi, I'm using Lepton 3.5 and purethermal 2, but my result temperature is around 100 (it should be close to 37,right?), and when I use the notebook(https://github.com/tomek-l/ai-thermometer/blob/master/docs/notebooks/Getting%20Started%20with%20Cameras.ipynb), the raw data for each pixel is around 8000 and after Kelvin to Celsius the result is around -195.
Do you have any idea about this?  Thanks.

tomek-l, 2020-08-31:
That's very weird. For me the raw values are (e.g. ~30315 for room at ~30deg C). The values being in 8000 range imply that the data is indeed in GRAY16_LE 16-bit format.

1. I presume that the following:
```shell
!gst-launch-1.0 v4l2src device=/dev/video1 ! video/x-raw,format=GRAY16_LE ! videoconvert ! ximagesink
```
gives you very dark (near black) image?

2. Could you send me an example png file from ```get_ir_image()``` so I can take a look?

You can use the following command to save a file:
```shell
!gst-launch-1.0 v4l2src device=/dev/video1 num-buffers=1 ! video/x-raw,format=GRAY16_LE ! videoconvert ! pngenc ! filesink location=/tmp/ir.png
```
It should be in ```/tmp/ir.png```

3. You are using jetson nano, right? Which version of jetpack?  What's the output of the following command?
```shell
gst-inspect-1.0 pngenc
```

weshao, 2020-09-01:
@tomek-l Thanks for the reply.
This is the running image:
[image]

1. the image is not very dark, it's gray.
[image]

2.The image
[image]

3. Yes, I'm using Jetpack 4.4 for Jetson Nano.
The output is:

```
Factory Details:
  Rank                     primary (256)
  Long-name                PNG image encoder
  Klass                    Codec/Encoder/Image
  Description              Encode a video frame to a .png image
  Author                   Jeremy SIMON <jsimon13@yahoo.fr>

Plugin Details:
  Name                     png
  Description              PNG plugin library
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstpng.so
  Version                  1.14.5
  License                  LGPL
  Source module            gst-plugins-good
  Source release date      2019-05-29
  Binary package           GStreamer Good Plugins (Ubuntu)
  Origin URL               https://launchpad.net/distros/ubuntu/+source/gst-plugins-good1.0

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstVideoEncoder
                         +----GstPngEnc

Implemented Interfaces:
  GstPreset

Pad Templates:
  SINK template: 'sink'
    Availability: Always
    Capabilities:
      video/x-raw
                 format: { (string)RGBA, (string)RGB, (string)GRAY8, (string)GRAY16_BE }
                  width: [ 1, 2147483647 ]
                 height: [ 1, 2147483647 ]
              framerate: [ 0/1, 2147483647/1 ]
  
  SRC template: 'src'
    Availability: Always
    Capabilities:
      image/png
                  width: [ 16, 1000000 ]
                 height: [ 16, 1000000 ]
              framerate: [ 0/1, 2147483647/1 ]

Element has no clocking capabilities.
Element has no URI handling capabilities.

Pads:
  SINK: 'sink'
    Pad Template: 'sink'
  SRC: 'src'
    Pad Template: 'src'

Element Properties:
  name                : The name of the object
                        flags: readable, writable
                        String. Default: "pngenc0"
  parent              : The parent of the object
                        flags: readable, writable
                        Object of type "GstObject"
  qos                 : Handle Quality-of-Service events from downstream
                        flags: readable, writable
                        Boolean. Default: false
  snapshot            : Send EOS after encoding a frame, useful for snapshots
                        flags: readable, writable
                        Boolean. Default: false
  compression-level   : PNG compression level
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 9 Default: 6 

```

tomek-l, 2020-09-09:
Hi @weshao 
Sorry it took me so long to get back to you! I couldn't figure it out, but I ran into a similar issue tonight on different hardware.

You're likely trying to use [Lepton 3.0](https://www.digikey.com/product-detail/en/flir-lepton/500-0726-01/500-0726-01-ND/6163867) (S/N: 500-0726-01), instead of [Lepton 3.5](https://www.digikey.com/product-detail/en/flir-lepton/500-0771-01/500-0771-01-ND/7606616) (S/N: 500-0771-01) module. 
They look the same and it's easy to get them confused. 

Unfortunately, Lepton 3.0 doesn't support radiometry, which is required to obtain the temperature values.

weshao, 2020-09-11:
Hi @tomek-l
Yes, you are right, I checked and found that I am using Lepton 3.0. 
Thank you for your answer :)

closed 2020-09-11 by weshao