tomasz-lewicki/ai-thermometer #6: Running with web cam and IR cam, not accurate temperature reading

nickjrz, 2020-09-14:
Hi,

First of all, thanks for sharing the source code for this project, very helpful. I have been trying to run this code with a web cam instead of using the raspberry pi cam. I have it working with display and face detection, the only issue I am running into is that the temperature readings are not accurate. Would you have any idea what could be causing this behavior?

tomek-l, 2020-09-15:
Hi @nickjrz ,

Commit 9d754093d3fcf76fe9ce421dad255f20c7e582ec introduced a bug which led to incorrect readings. I just pushed 097c9c904fc412f0e25225afdf44eb3ec60b2f82 which fixed that. Let me know if the readings are better now :slightly_smiling_face:

nickjrz, 2020-09-15:
Thanks for your quick reply, I will try it today and will keep you updated.

nickjrz, 2020-09-15:
So I updated the files and I am still getting readings of 62550 celsius. I found a formula for the lepton thermal cameras temp  = slope * (raw -8192) + ambTemp  `https://groups.google.com/g/flir-lepton/c/T2NFRkeMyR4?pli=1` maybe this could be used along the calculations on code

tomek-l, 2020-09-15:
Hi,

Is it possible that you're using [Lepton 3.0](https://www.digikey.com/product-detail/en/flir-lepton/500-0726-01/500-0726-01-ND/6163867) (S/N: 500-0726-01), instead of [Lepton 3.5](https://www.digikey.com/product-detail/en/flir-lepton/500-0771-01/500-0771-01-ND/7606616) (S/N: 500-0771-01) module?

nickjrz, 2020-09-15:
So I have the lepton 3.0 and 2.0, as of now I am using the 2.0 model. I think I only need to add some calibration coefficient in the calculation in order for it to get readings properly.

tomek-l, 2020-09-15:
In order to get the temperature values out-of-the-box you'd have to have one of the **radiometric** lepton modules (e.g. Lepton 3.5). They come pre-calibrated and stream temperature values in Kelvin.

nickjrz, 2020-09-15:
Okay thanks, I guess I just need to figure out how to calibrate it. I tried ordering the 3.5 one but they were out of stock and only had 2.0 so I will keep you updated if I make any progress on this model.

nanogear, 2020-09-25:
@nickjrz try to buy the lepton directly to Groupgets or Digikey. Both have the radiometric 3.5 in stock

muralikrishna416, 2020-10-06:
Hai, i want to know small information like without IR Thermal sensor, the camera will open or will not open through the code which you have provided.

tomek-l, 2020-10-06:
Without the Lepton camera, the program will just indefinitely wait for IR
frames and display no output.

You can run just the RGB object detection with the script here:
https://www.github.com/tomek-l/ai-thermometer/tree/master/vis%2Fssd%2Fexample_detect_face_caffe_resnet.py

nickjrz, 2020-10-07:
I am currently waiting for a lepton 3.5 (digikey) to come in, as of now I am still trying to get the 3.0 to work with this application with some sort of calibration.

nickjrz, 2020-10-08:
I just got the lepton 3.5 and tried it out altogether but readings seem to be a little off. I am getting an average temperature of 32 Celsius. Do I need to let it run for a while for it to warm up?

tomek-l, 2021-02-27:
The freshly merged PR #31 adds:
- An option for adding an external blackbody reference
- Reports the average + maximal + 90th percentile temperature of the face, not just a single value.

You should pull the newest master give it a try :slightly_smiling_face: