tomasz-lewicki/ai-thermometer #7: where can i get thermal from ur file

ROK-KJ, 2020-09-16:
i want get a putext and thermal(degree) information ur file..
but i cannot find it
please help me ;)

tomek-l, 2020-09-16:
Hi @ROK-KJ ,

Would you like a sample image file (e.g. in png format) from thermal camera?
Not sure if I understood correctly :)

ROK-KJ, 2020-11-09:
hello again.
main.py is for Version 0.3: ResNet-10 Facial Detection + FLIR Lepton 3.5 temp. measurement (~17 FPS w/ GPU accel.)??
i want to know how did you run this face detect by jetson GPU.

tomek-l, 2021-02-27:
Hi @ROK-KJ ,
Sorry to take a while to respond. The current version which supports:
* SSD with ResNet-10 backbone (see code [here](https://github.com/tomek-l/retinaface/tree/5f1f892d80a44a01f46b03ce767ce03417050ae5) )
* RetinaNet with Mobilenet backbone (see code [here](https://github.com/tomek-l/retinaface/tree/5f1f892d80a44a01f46b03ce767ce03417050ae5) )