tomasz-lewicki/ai-thermometer #8: Incorrect Temperature values

jazz-ai, 2020-09-18:
It always shows human body temperature around 32-34 degree Celsius but actual temperature is always around 36-37 degree. How can I get accurate values? Are you using any calibration of Lepton 3.5 camera?

tomek-l, 2020-09-18:
Hi @jazz-ai 

1) What lens are you using? You're probably using a different field of view, hence the mismatch. I plan to update the code with support for different FOVs sometime.
2) Keep in mind that we're measuring the average skin surface temperature, which will be significantly lower (~33-36 deg C) than body core temperature (36-37 deg C).

razilla, 2020-11-16:
is there a way to calibrate the difference manually im have the same temperature issue (90.0 def F) the lens i am using is flir 3.5

nanogear, 2020-12-16:
I am having the same issue... @jazz-ai do you solved the issue?

tomek-l, 2021-02-27:
1. PR #31 adds an optional external blackbody measurement. That improves the accuracy dramatically.
2. Now I display much more comprehensive info about the facial temperature (avg+max+90th percentile), not just a single number.
3. There was briefly a camera calibration bug, which might have caused temperature readouts from the incorrect area in the image (fixed in e2fc37707c8f44a0c80dae196fccdb94971e0f23)

Anyway, one or more of those should solve the issue :smile: