tomasz-lewicki/ai-thermometer #11: UVC error

razilla, 2020-10-11:
Hey Tomasz

Thanks for putting this out for people its a really cool project bro. I'm having trouble finding the video; I get the following error. Any idea trouble where I messed up? I'm thinking it may be how cameras are lined up. I 3d printed the case you gave but I noticed its slightly different than what you showed in the blog post. I attached a picture of how it look on my end was hoping if you thought that was maybe the problem. Also I 3d printed an Iron Man Helmet as a case I think you would appreciate. I attached that as well.

Thanks again,
Raza

[image]
[image]
[image]

 /ai-thermometer$ python3 main.py
Loading weights from file...
Weights loaded!
Running first net inference...
Detector initialized!
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0
   Camera mode  = 0
   Output Stream W = 3264 H = 2464
   seconds to Run    = 0
   Frame Rate = 21.000000
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0] global /tmp/build_opencv/opencv/modules/videoio/src/cap_gstreamer.cpp (935) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
uvc_open error -3

glbt, 2020-10-12:
This is permissions on the camera I believe, try it with sudo and see if it works :)

razilla, 2020-10-12:
Not getting the frames

raza@raza-desktop:~/ai-thermometer$ sudo python3 main.py
Loading weights from file...
Weights loaded!
Running first net inference...
Detector initialized!
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain rang                                                                          e min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain rang                                                                          e min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain rang                                                                          e min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range                                                                           min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range                                                                           min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0
   Camera mode  = 0
   Output Stream W = 3264 H = 2464
   seconds to Run    = 0
   Frame Rate = 21.000000
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0] global /tmp/build_opencv/opencv/modules/videoio/src/cap_gstreamer.cpp                                                                           (935) open OpenCV | GStreamer warning: Cannot query video position: status=0, va                                                                          lue=-1, duration=-1
device opened!
Version gpp: 0.0.0 dsp: 0.0.0
FLIR part #: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\                                                                          x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
FLIR serial #: b'\x00\x00\x00\x00\x00\x00\x00\x00'
format: b'UYVY'
  frame 80x60 @ 9fps
format: b'Y16 '
  frame 80x60 @ 9fps
  frame 80x63 @ 9fps
format: b'Y8  '
  frame 80x60 @ 9fps
format: b'RGBP'
  frame 80x60 @ 9fps
format: b'}\xeb6\xe4'
  frame 80x60 @ 9fps
Estimated / selected altsetting bandwith : 13 / 642.
No protocol specified
Unable to init server: Could not connect: Connection refused

(python3:8113): Gtk-WARNING **: 11:53:00.221: cannot open display: :0
CONSUMER: Done Success
WARN: no frame
WARN: no frame
WARN: no frame
WARN: no frame
WARN: no frame
WARN: no frame
WARN: no frame
WARN: no frame
WARN: no frame
WARN: no frame
WARN: no frame
WARNING Argus: 5 client objects still exist during shutdown:
        547884820536 (0x7f04003b88)
        547891253568 (0x7f040018c0)
        547891253728 (0x7f04001960)
        547891258416 (0x7f04001ac0)
        547891259712 (0x7f04003a70)

tomek-l, 2020-10-14:
Hi @razilla !

As @glbt  said, It looks like an issue with r/w permissions for the USB device.

I posted a solution in [readme](https://github.com/tomek-l/ai-thermometer#make-sure-your-user-has-rw-access-to-the-camera-device) under  "Common Issues"

If you do the following, you should be able to run this code without `sudo`.
```shell
echo 'SUBSYSTEM=="usb",  ENV{DEVTYPE}=="usb_device", GROUP="plugdev", MODE="0664"' | sudo tee /etc/udev/rules.d/10-libuvc.rules 
sudo udevadm trigger
```
Let me know if the other issue persists with the new udev rules & without sudo :slightly_smiling_face:

tomek-l, 2020-10-14:
If you refer to the post on hackster.io, I posted it a while back and probably updated the enclosure design since then :) 
Also, cool mask! :smile:

deantheiceman, 2020-10-19:
Hi @tomek-l reallly nice work. I have the same problem, getting 

```
waiting for rgb frames
Warn: no frame
```
I also tried running as root still got the same problem

Any suggestions?

tomek-l, 2020-10-19:
Hi @deantheiceman !

Looks like the gstreamer pipeline got initialized correctly, but we're not getting any RGB frames upon `.read()`.

This can occur if something crashes on `nvargus` side. Most of the time restarting it should solve the problem:
```
sudo systemctl restart nvargus-daemon
```

Let me know if this helped! :slightly_smiling_face:

deantheiceman, 2020-10-20:
Hi @tomek-l thanks for the reply. I tried rebooting and also restarting the service with no luck, 

```
CONSUMER: Producer has connected; continuing.
[ WARN:0] global /tmp/build_opencv/opencv/modules/videoio/src/cap_gstreamer.cpp (935) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
uvc_open error -3
```

Although the error message seems to be slightly different

nanogear, 2020-10-26:
> Hey Tomasz
>
> Thanks for putting this out for people its a really cool project bro. I'm having trouble finding the video; I get the following error. Any idea trouble where I messed up? I'm thinking it [...]

So, just to get this clear...the position of the cameras as Razilla have them are ok to work with?

razilla, 2020-10-29:
@nanogear 
Yes the positions work where I put them. You should see two different screens when you get it running so the positioning can vary slightly is what I've noticed.
The  screens actually seem to operate independently from one another. i.e. you can have two people in infrared and only one in the RGB or vice versa but you really have to be trying to do that.

[image]

closed 2020-10-29 by razilla

nanogear, 2020-11-12:
> Hi @tomek-l thanks for the reply. I tried rebooting and also restarting the service with no luck,
> 
> ```
> CONSUMER: Producer has connected; continuing.
> [ WARN:0] global /tmp/build_opencv/opencv/modules/videoio/src/cap_gstreamer.cpp (935) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
> uvc_open error -3
> ```
> 
> Although the error message seems to be slightly different
> 
> _EDIT_: Got it working, thanks so much

Can you explain me how you fix it please? 

When applying the "dirty solution" I get:  ENV{DEVTYPE}=="usb_device,: command not found