tomasz-lewicki/ai-thermometer #12: Version

glbt, 2020-10-12:
Hey,

Thanks for this project, I've had great fun working on this, an area I am really new too.  A quick question which may be worth updating in the notes, which version is this?  It appears to be v.01 with the simple box detection?  Is that correct?  I'm really interested in the eye detection model 👍 

Many thanks,

David

tomek-l, 2020-10-14:
Hi there!

Right now I'm using an SSD classifier trained on FDDB dataset. v0.1 was trained on COCO.
You can browse the repository state from version 0.2 [here](https://github.com/tomek-l/ai-thermometer/tree/39287df98c08af0f10fa21471c385894d6a81ad4). Eye detection happens exactly [here](https://github.com/tomek-l/ai-thermometer/blob/39287df98c08af0f10fa21471c385894d6a81ad4/main.py#L96) :slightly_smiling_face: 

I have recently modularized the architecture and I'm planning to add the older facial/object detectors, since a lot of people ask me about them.

glbt, 2020-10-16:
Thanks so much, I'm so experienced in nearly every other part of IT apart from python, I'm on a crash course!  After many other languages it's really nicely laid out and super east to debug, I can see why it's gained so much traction.  I'll contribute once I'm a bit more proficient.

closed 2021-02-27 by tomek-l