tomasz-lewicki/ai-thermometer #13: Interface for deployment

deantheiceman, 2020-10-22:
Hi,

Thanks so much for your work this is really interesting. 

I was wondering what kind of interface have you used for actual deployment. I saw you were working on a web interface, does that work? For us to put on a screen near the camera display?

Would be nice to have like an audio alert which we could play maybe through the web interface if someone is detected above the desired range.

Also curious if you're using the flir 3.5 as well, do you need any calibration? My detection seems to be a little bit off

Thanks again

tomek-l, 2021-01-14:
Hi @deantheiceman ! Sorry for getting to you so late.
1. I currently use a regular HDMI output & a simple window OpenCV based interface under X-server.
2. I did work on a http-based interface a while back (see b6711bc30a28c0b60db8c1ea6a334ba20b582e1b), but never merged it to master. 

Do you have any suggestions w.r.t how the interface should be realized?

tomek-l, 2021-01-14:
As far as the second part of your question goes:
Te IR camera has an inherent temperature drift that will affect the readings. That's why I plan to implement calibration with an external blackbody reference in an upcoming version of the software.
I set up issue #24 to hear suggestions on how this should be realized, feel free to comment :slightly_smiling_face:

deantheiceman, 2021-01-16:
hi @tomek-l thanks for the reply.

Something like what they show here would be really useful https://www.tintahijau.com/images/2019-2/001_Hikvision.jpg

We setup your project in our small office, we have a TV display people as they come in. But it'll be useful if we could track if someone goes above the threshold and then save that image and throws an alarm if they are above. In the other examples they track a picture of each unique person that came in.

Thanks again for the great work. Really wish I could help

deantheiceman, 2021-01-16:
Another nice way commercial projects have implemented it is like this 
[image]

So the logic is like:

- if face detected
- if below temp show green lights with the temp
- if above take picture, throw alarm

Maybe we can do it with the http interface, but my pthon skills are really bad :(

deantheiceman, 2021-01-31:
Hi @tomek-l sorry to bump this, do you think you will merge the web interface into master?

tomek-l, 2021-02-03:
Hi! 

Let me give a little bit of background on that.
I'm looking for something that would fulfill the following 4 criteria:
* no issues in major browsers
* low enough latency (2s maximum)
* decent framerate (~5FPS at minimum)
* streaming to multiple clients without impeding the performance of the main app. (at least 3 devices)

The closest I got to that goal was by simply streaming a jpeg sequence to the browser. 
That's available on the `http_streaming` branch (b6711bc) and works out perfectly for a single device.

Unfortunately, the per-client memory footprint is high, and I'm unable to handle >2 clients without impact on the main app.
I'm looking for a better solution, and dragging my feet on merging that to master for that reason.

tomek-l, 2021-02-03:
You know @deantheiceman, now that I think about it more, I was streaming a full `768x1024px` frame to the browser. 
Maybe I should give it a try with something smaller and see what happens.

deantheiceman, 2021-02-08:
Thanks, happy to test if you get anything going. I noticed the screenshot rules also really impact performance, disabling that got me much better performance. Would be useful if we could snapshot when they trigger the increased temperature

tomek-l, 2021-02-27:
Hi @deantheiceman !

Feel free the newest version. It has new and improved interface.