tomasz-lewicki/ai-thermometer #16: bash: ENV{DEVTYPE}=="usb_device,: command not found

nanogear, 2020-11-12:
Thank you again for all the help!!!

Can you explain me how you fix it please? When applying the "dirty solution" I get:

bash: ENV{DEVTYPE}=="usb_device,: command not found

What I am doing wrong?

Best regards!!!

tomek-l, 2020-11-12:
Hi,
Looks like something is wrong with quotation marks. The whole command is:
```shell
echo 'SUBSYSTEM=="usb",  ENV{DEVTYPE}=="usb_device", GROUP="plugdev", MODE="0664"' | sudo tee /etc/udev/rules.d/10-libuvc.rules 
```

Are you able to do the following without an error?
```
echo 'SUBSYSTEM=="usb",  ENV{DEVTYPE}=="usb_device", GROUP="plugdev", MODE="0664"'
```

It should just print the line to the terminal.

Let me know if it helped :slightly_smiling_face:

nanogear, 2020-11-12:
Ohhhh, silly me!!! it was a wrong quotation mark!! Code works fantastic!!!!!!

Another silly question that is just mere curiosity... do you know why it took like 6.5 hours to build/compile/install the OpenCV 4.4.0 package? hehe

Thank you very much!!!!

tomek-l, 2020-11-12:
I'm glad it worked for you :slightly_smiling_face: 

Well, OpenCV is a pretty big project with lots of modules and Jetson Nano is a small computer :slightly_smiling_face: 

You can speed up the compilation a little bit by increasing the number of jobs that make will use here:
https://github.com/mdegans/nano_build_opencv/blob/0832a850f33a64e3c14de065ed205db863b25f0e/build_opencv.sh#L16 

But then you might run out of the 4GB of memory, so [adding more swap](https://github.com/JetsonHacksNano/resizeSwapMemory) is recommended.

nanogear, 2020-11-12:
I understand now... I appreciate so much your help!! Thank you!!!

closed 2020-11-13 by nanogear