tomasz-lewicki/ai-thermometer #17: Syntax error

nanogear, 2020-11-16:
Hello, need some advice please.. why I am getting syntax error when executing Capture Images.ipynb example?

Thank you and Regards!!!

`!gst-launch-1.0 v4l2src device=/dev/video1 num-buffers=1 ! video/x-raw,format=GRAY16_LE ! videoconvert ! pngenc ! filesink location="$save_path"`

tomek-l, 2020-11-16:
Hi nanogear!

Are you executing this line within a notebook or in a shell?
1) If in the notebook, could you show the error you're getting?
2) If in the shell, remove the preceding '!' from the command and provide a
specific file location, like so:
gst-launch-1.0 v4l2src device=/dev/video1 num-buffers=1 !
video/x-raw,format=GRAY16_LE ! videoconvert ! pngenc ! filesink
location=/tmp/image.png

Let me know if this helps!

Tomasz

nanogear, 2020-11-17:
Thank you Thomasz 

It worked now. Executing in the notebook.

closed 2020-11-17 by nanogear