tomasz-lewicki/ai-thermometer #18: (NOT AN ISSUE) Neopixel to indicate "fever" in a detection

nanogear, 2020-11-24:
I am trying to use Neopixels on this project,. Recently Adafruit coded some libraries for the Neopixels that are useful in the Jetson Nano using CircuitPython. It works pretty nice...

I want to make the Neopixel stay ON with green colour when the Jetson Nano be in fever screening mode and when a 37.5 C temperature is detected, the Neopixel changes the colour to red for example... I need some help to incorporate it to the Tomasz project.

Here a simple tutorial to make Neopixels work with the Jetson Nano:

Install Pip3 if not already installed:
`sudo apt-get install python3-pip`

Reboot

Install Adafruit-Blinka 5.8.0:
`pip3 install Adafruit-Blinka`

Enable SPI on Jetson Nano:
`sudo /opt/nvidia/jetson-io/jetson-io.py`

- Select Configure 40-pin expansion header at the bottom.

- Select spi1 (19, 21, 23, 24, 26) and then select Back 

- Finally select Save and reboot to reconfigure pins. This will create a config file and reboot the Jetson Nano.

After the Nano boots up again, verify you have the I2C devices with the command:
`ls /dev/i2c* /dev/spi*`

You can test to see what I2C addresses are connected by running `sudo i2cdetect -r -y 0` (on pins 27/28) or `sudo  i2cdetect -r -y 1` (on pins 3/5)

Then, we need to set user permissions. In order to use the Jetson GPIO Library, the correct user permissions/groups must be set first. Start by creating a new gpio user group:

`sudo groupadd -f -r gpio`
`sudo usermod -a -G gpio pi`

Clone the Repo and copy the rules:

`cd ~`
`git clone https://github.com/NVIDIA/jetson-gpio.git`
`sudo cp ~/jetson-gpio/lib/python/Jetson/GPIO/99-gpio.rules /etc/udev/rules.d`

_This section was originally written for releases before JetPack 4.3. For JetPack 4.3, the udev rules are in: /lib/udev/rulesd/60-jetson-gpio-common.rules. This should mean that you don’t have to copy the file anymore with JetPack 4.3 and above._ 

 _Also, it appears as if the default is for users to be included in the gpio group. You can check this by:_
`$ groups`

Reboot

Update just in case if a library is missing:
`sudo pip3 install --upgrade adafruit_blinka`

Install  CircuitPython NeoPixel_SPI library;:

`pip3 install adafruit-circuitpython-neopixel-spi`

Reboot again and that´s all!!!

Connect "Din" of your Neopixel to the SPI1_MOSI pin of your Jetson Nano(pin 19 of the expansion header), Gnd to Gnd and 5v to 5v.

try this example:

`import board`
`import neopixel_spi`

`pixels = neopixel_spi.NeoPixel_SPI(board.SPI(), 10)`
`pixels.fill(0xff0000)` 

You can check the Hex colors to use whatever colour you want by using this web page:
[https://www.color-hex.com/](url)
-------------------------------------------------------------------------------------------------------------------

tomek-l, 2020-11-24:
Hi @nanogear !

Neat idea! I thought about it a little bit before, but never implemented it :slightly_smiling_face:
The only way of controlling neopixels I knew about was by using a bit-banged protocol for the ws2812, but that required 100% of one CPU core, so I gave up on this. Looks like `NeoPixel_SPI` library lets you use `SPI` hardware, without hogging the CPU :+1: 

Let me take a look at the library and get back to you! :slightly_smiling_face:

tomek-l, 2021-02-23:
Started working on this feature in c7983eada39b24c398912b355df1523b5348e0e2 (see [image]).

Will add modified 3D printed enclosure that supports neopixels soon.

nanogear, 2021-02-26:
Awesome Tomasz, thank you very much for your kind work!!