tomasz-lewicki/ai-thermometer #19: White box on RGB Camera out of sync with IR Camera

aaronf99, 2020-12-04:
Hello,
I have copied your project directly, using the same casing. After running the program, I can see that the frame of reference is off. Is there a fix for this?
[image]
[image]
[image]
[image]

tomek-l, 2021-01-14:
Hi there, @aaronf99 !

Yup, that's a bug. I was experimenting with a different RGB camera with a wider FOV and merged code to master, but the publicly available hardware for that is not yet updated. In the future, the system will support multiple cameras with different FOVs. I will probably also start a "stable" branch to avoid issues like that in the future.

aaronf99, 2021-01-14:
Thanks for the reply @tomek-l 

That makes sense, I am using the Raspberry Pi Camera Module V2.
Is there a quick fix I can do for this as the area that I currently work in could really do with this right now.

Many Thanks,
Aaron

tomek-l, 2021-01-24:
Hi @aaronf99!

e2fc37707c8f44a0c80dae196fccdb94971e0f23 introduces a hotfix for this bug. Checkout the newest master, and let me know if that looks better. 

The alignment could still be a bit off at a distance. The reason is the way the sensor is attached to Raspberry Pi CM V2.1 PCB (double sided tape). Due to that the RGB sensor will be mounted at a slightly different angle in different units (that's why I'm experimenting with a different cameras in the first place). I will also work out support for cameras with different FOVs in a future version.

closed 2021-01-27 by aaronf99