tomasz-lewicki/ai-thermometer #21: Wrong viewing mode on the FLIR-Lepton

Lord-Lukas-1, 2020-12-29:
I copied your code and am running it on the developer kit. Unfortunately the FLIR Lepton shows the pictures only in red and grey like in your video with the red channel. How can i change the output of the FLIR Lepton to a range from blue to red like in your example?

nanogear, 2021-01-05:
In main.py add the next code:
`ir_arr_n = cv2.applyColorMap(ir_arr_n, cv2.COLORMAP_JET)` where it is defined `def apply_cmap`

Just still don´t understand how you change blue color to gray and modify the other ones to look like in the video.... I want to know how...

Lord-Lukas-1, 2021-01-12:
Thank you for the answer.

I added the code but unfortunately it didn't work. Do you have any idea what else could be the problem?

I haven't made any changes in the code after copying  it from here except adding your line. So I have no idea what is causing the problem.

tomek-l, 2021-02-27:
@Lord-Lukas-1 you should give the newest version a try, it includes improved colormaps.