tomasz-lewicki/ai-thermometer #23: Calibrating Lepton 3.5 for Outdoor vs Indoor

rgpettit, 2021-01-11:
Has anyone run into calibration issues when using this outdoors - particularly in winter?  The system works perfectly inside with ambient temperatures of 66-72F.  However, trying to use it outdoors <45F produces really drastic readings on the facial detections - typically over 1000F!

I've tried leaving the system outside but am still having the same problems.  I'm a track & field coach and am trying to set this up to screen athletes so I need it to work outside.  Any suggestions would be greatly appreciated.

Thanks!

rgpettit, 2021-01-13:
UPDATE -  it's not the cold temperatures but rather sunlight that seem to be the problem.  Works perfectly in my cold garage but as soon as it's in the sun, reading go crazy.  It's not even facing the sun directly - just the presence of sunlight affects it.

nanogear, 2021-01-13:
Probably this is your problem... emissivity:

https://www.flir.eu/discover/professional-tools/how-does-emissivity-affect-thermal-imaging/

tomek-l, 2021-01-14:
Hi @rgpettit !

That's a really great application and I hope that I can be of help :slightly_smiling_face: 

Lepton 3.5 is capable of sensing up to 150C in low-gain mode, and 450C in high-gain mode.
It's therefore not capable of producing a (valid) reading reading of 537C (1000F), so I doubt that it's an environmental factor.
The bug is likely caused by:
- a bug on my side in processing
- you're running a different version of lepton hardware than I do

1. Would you be able to share some visuals illustrating the problem? Could be a screenshot of the AI Themometer UI, or the output of this `gstreamer` command: `gst-launch-1.0 v4l2src device=/dev/video1 ! video/x-raw,format=UYVY ! videoscale ! video/x-raw,width=800,height=600 ! videoconvert ! ximagesink`
2. Could you confirm which lepton version are you running (3.0, 3.5, radiometric/non-radiometric)?

rgpettit, 2021-01-14:
Thanks so much for your help!  I am using the radiometric Lepton 3.5 from DigiKey.  Images attached from both the UI and gstreamer.  "Normal" images were taken in front of a windowed door with full sun in the background.  "Sun" images were taken standing in the same spot but with that door open.  I get the same "normal" results anywhere indoors and the same "sun" results anywhere outside during the day.

Hope this makes sense.  As you might notice from the UI image, I'm working with one-arm after injuring myself in T&F :-)

[image]
[image]
[image]
[image]

nanogear, 2021-01-14:
Hello,

Hope everybody is doing well. I tried to do the same today. Put the system in front of an open window(seeing the exterior light but not directly to the sun) and I am having the same results as @rgpettit 

I am using the Flir Lepton 3.5(radiometric). Indoor works pretty good.

[image]

nanogear, 2021-01-23:
@rgpettit Hi, have you found a solution to the issue? Regards

rgpettit, 2021-01-23:
Nothing yet, but I haven't had much time to look into it the last week or so.  @tomek-l, have you discovered anything either?

tomek-l, 2021-01-24:
Hi there! 

I was able to reproduce the bug. It was due to uint16 type underflow during conversion from Kelvin to Celsius, and
328b29349f4f3e57665feb9bb87b527504821fd4 should fix this. 
I knew that Lepton 3.5 has (-10C to 400C) range, but I never really tested that -10C to 0C part of it in California :slightly_smiling_face: 
Checkout the newest `master` branch, and it should work!

rgpettit, 2021-01-24:
Excellent!  Great find - works perfectly for me now.  Thanks!

closed 2021-01-24 by rgpettit

nanogear, 2021-01-25:
It worked for me too. Thank you very much!!!!