tomasz-lewicki/ai-thermometer #26: Storage on Jetson Nano Full

aaronf99, 2021-01-27:
Hello, 
Thanks for all the help so far, I have an issue whereby the storage on my Jetson Nano fills up after a few hours. I have no value in saving the frames. I only use this system for a live feed on a monitor.
Is there a way I can stop it from saving frames? 
Thanks

tomek-l, 2021-01-27:
Hi @aaronf99,

1) Does the application crash? If so, I should treat it as a bug.
2) To answer your question, you have to set `SAVE_FRAMES` to `False` in [config](https://github.com/tomek-l/ai-thermometer/blob/e2fc37707c8f44a0c80dae196fccdb94971e0f23/main.py#L28):

aaronf99, 2021-01-27:
Hello,

So the Jetsons external storage would fill up, upon reboot I get the NVIDIA Logo then it crashes, once I clear the memory it works again.
Thanks for the response, will apply the edit to the config file upon reseting the Nano.

Thanks

tomek-l, 2021-01-27:
Okay, it might be worth changing the default behavior of the system - i.e. before each save, we could check if there's enough storage left. This could be performed with a single check like this:
```
if shutil.disk_usage(LOG_DIR).free < 1e9 # less than 1GB
```

closed 2021-01-29 by aaronf99