tomasz-lewicki/ai-thermometer #27: Adding support for SPI lepton board: let me know what you think!

tomek-l, 2021-01-29:
I'm leaning towards using SPI directly (rather than USB) for interfacing Lepton 3.5 in a future hardware revision.

* It would bring down the cost of the device (the GroupGets Purethermal board costs ~100USD).
* It would mitigate the need for bulky USB connectors inside the case.
* The transition should be relatively easy. The [pylepton](https://github.com/groupgets/pylepton) package could be used to interface lepton module.

If you're using AI Thermometer and have any thoughts on this change, feel free to comment below.

CristinaPosada, 2021-02-16:
Hi! This would be very interesting! 
I do not have the Purethermal, I have the breakoutboard V2 and I would like to use ai-thermometer

123Dante123, 2021-04-02:
Hey Tomasz, are you still planning to add support for the SPI lepton board?

Skylakc, 2024-07-16:
Hello Tomasz! Are u still developing the SPI feature?