tomasz-lewicki/ai-thermometer #28: [enhancement] Improve consistency of face detection when wearing a mask and on moving target

nickjrz, 2021-02-04:
Hi,

Just wanted to propose and to see if there is anyway to improve consistency and accuracy of the face detection when people are wearing masks or when the target is moving. Maybe training the model with people wearing masks and not wearing masks.

Let me know what you think and if it is doable.

Nick

tomek-l, 2021-02-27:
Hi @nickjrz ,

Check out the most up-to-date version.
The facial detection performance with mask on has vastly improved :slightly_smiling_face:

nickjrz, 2021-03-02:
Thanks for getting back to me on this. The most up-to-date version works great with masks and moving targets as well. 

My hot fix for this was re-training the resnet 10 model with a dataset that included people with and without masks. Zooming in on the RGB cam live stream also improves drastically the face detection on moving targets using the resnet model. Only thing I was not able to figure out was how to zoom in on the IR cam feed to make the shifting work properly.