tomasz-lewicki/ai-thermometer #30: Camera connection?

Thanh-Binh, 2021-02-09:
thanks for sharing us your work.
I understood that you use NVidia Nano Dev Kit with 2 cameras: Visible and Lepton 3.5.
Could you please share me information which visible camera you use and how to connect all camera with Nano? via MIPI-CS2 or SPI?
Thanks

tomek-l, 2021-02-27:
Hi @Thanh-Binh ! 

I use:
- Raspberry Pi Camera Module V2.1 connected via MIPI-CSI2
- Lepton 3.5 connected via PureThermal2 usb board

A while ago I wrote a [post](https://www.hackster.io/tomasz-lewicki/ai-thermometer-2bacb4) you might find helpful.

Thanh-Binh, 2021-02-27:
Hi Atomen
Thanks for your response!
1. I do not know about Jetson Nano. How can you compile and run opencv face and eye detection on it?

2. how can you calculate body core Temperatur based on face skin temperature?

3. do you plan to use DNN to detect face and facial landmarks directly from thermal images?

Thanks 

Gesendet mit der mobilen Mail App