tomasz-lewicki/ai-thermometer #33: Error occur in ai-thermometer!! Please Guide me to resolve this problem

Henery1, 2021-03-13:
File "main.py", line 13, in <module>
    from rgb import RGBThread
  File "/home/thermal/ai-thermometer/rgb/__init__.py", line 1, in <module>
    from .rgb_thread import *
  File "/home/thermal/ai-thermometer/rgb/rgb_thread.py", line 8, in <module>
    from .retinaface import RetinaFaceDetector
  File "/home/thermal/ai-thermometer/rgb/retinaface/__init__.py", line 1, in <module>
    from .detector import *
  File "/home/thermal/ai-thermometer/rgb/retinaface/detector.py", line 10, in <module>
    from .models.retinaface import RetinaFace
  File "/home/thermal/ai-thermometer/rgb/retinaface/models/retinaface.py", line 5, in <module>
    import torchvision.models.detection.backbone_utils as backbone_utils
ModuleNotFoundError: No module named 'torchvision.models.detection'

tomek-l, 2021-03-13:
Hi @Henery1 

This looks like a mismatch between `torchvision` versions.
It was likely caused by an issue in installation instructions.
I'll look into this!

tomek-l, 2021-03-13:
@Henery1 

What version of `torch` and `torchvision` are you using?
You can check that by inspecting the output of the following commands:

```bash
pip3 show torchvision
pip3 show torch
```

Henery1, 2021-03-16:
Hi Tomek ! First of all Thanks you very much for your kindly respond. I used for thermal-view with purethermal-2 break out board and FLIR Lepton 3.5 and RGB for pi-camera v-2.1. Please Look at the torch and torchvision below:
For Torch 
Name: torch
Version: 1.8.0
Summary: Tensors and Dynamic neural networks in Python with strong GPU acceleration
Home-page: https://pytorch.org/
Author: PyTorch Team
Author-email: packages@pytorch.org
License: BSD-3
Location: /home/thermal/.local/lib/python3.6/site-packages
Requires: numpy, dataclasses, typing-extensions

For Torchvision
Name: torchvision
Version: 0.2.2.post3
Summary: image and video datasets and models for torch deep learning
Home-page: https://github.com/pytorch/vision
Author: PyTorch Core Team
Author-email: soumith@pytorch.org
License: BSD
Location: /home/thermal/.local/lib/python3.6/site-packages
Requires: six, pillow, numpy, torch

                                                                                                yours sincerely
                                                                                                         Henery T

Henery1, 2021-03-16:
Hi @tomek-l !! May I know Do you have an (ai-thermometer.img) file? If you have that kind of clone .img file Please share me or show me where i can download it. I tried many times and got the errors because of my few programming knowledge. Thanks a lot
                                                                                                      Yours sincerely
                                                                                                      Henery T

Henery1, 2021-03-18:
Hi @tomek-l <https://github.com/tomek-l> ! First of all Thanks you very
much for your kindly respond. I used for thermal-view with purethermal-2
break out board and FLIR Lepton 3.5 and pi-camera v-2.1 for RGB . Please
Look at the torch and torchvision below:
For Torch
Name: torch
Version: 1.8.0
Summary: Tensors and Dynamic neural networks in Python with strong GPU
acceleration
Home-page: https://pytorch.org/
Author: PyTorch Team
Author-email: ***@***.***
License: BSD-3
Location: /home/thermal/.local/lib/python3.6/site-packages
Requires: numpy, dataclasses, typing-extensions

For Torchvision
Name: torchvision
Version: 0.2.2.post3
Summary: image and video datasets and models for torch deep learning
Home-page: https://github.com/pytorch/vision
Author: PyTorch Core Team
Author-email: ***@***.***
License: BSD
Location: /home/thermal/.local/lib/python3.6/site-packages
Requires: six, pillow, numpy, torch

                     yours sincerely

                              Henery T

Henery1, 2021-03-18:
Hi @tomek-l <https://github.com/tomek-l> !! May I know Do you have an
(ai-thermometer.img) file? If you have that kind of clone .img file Please
share me or show me where i can download it. I tried many times and got the
errors because of my few programming knowledge. Thanks a lot
Yours sincerely
Henery T

Henery1, 2021-03-18:
PLease help me to get the ai-thermometer .img file. I really need your
project .img files . Please as soon as possible to share your ai
thermometer .img file. Thanks a lot

Henery1, 2021-03-22:
Hi @tomek-l 
I got that kind of error. Please guide me to get RGB and IR frame display.

(488) isPipelinePlaying OpenCV | GStreamer warning: unable to query pipeline state
[ WARN:1] global /tmp/build_opencv/opencv/modules/videoio/src/cap_gstreamer.cpp (1761) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module nvarguscamerasrc0 reported: TIMEOUT
GST_ARGUS: Cleaning up
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/thermal/ai-thermometer/rgb/rgb_thread.py", line 38, in run
    frame_small = cv2.resize(self._frame, (400, 300))
cv2.error: OpenCV(4.4.0) /tmp/build_opencv/opencv/modules/imgproc/src/resize.cpp:3929: error: (-215:Assertion failed) !ssize.empty() in function 'resize'

(960, 0)
Waiting for RGB frames
Waiting for RGB frames
Waiting for RGB frames
Waiting for RGB frames
Waiting for RGB frames
Waiting for RGB frames

tomek-l, 2021-03-22:
Hi @Henery1 ,
I'm glad you were able push it forward. Te second error is likely related to the gstreamer pipeline for RGB camera not having set up successfully. 

1. See if you can display camera 0 on the screen:
```shell
gst-launch-1.0 nvarguscamerasrc sensor_id=0 ! 'video/x-raw(memory:NVMM),width=3264, height=2464, framerate=21/1, format=NV12' ! nvvidconv flip-method=2 ! 'video/x-raw,width=960, height=720' ! nvvidconv ! nvegltransform ! nveglglessink -e
```
(or camera 1):
```shell 
gst-launch-1.0 nvarguscamerasrc sensor_id=1 ! 'video/x-raw(memory:NVMM),width=3264, height=2464, framerate=21/1, format=NV12' ! nvvidconv flip-method=2 ! 'video/x-raw,width=960, height=720' ! nvvidconv ! nvegltransform ! nveglglessink -e
```

2. If these commands don't succeed, give this a try:
```
sudo systemctl restart nvargus-daemon
```

Henery1, 2021-03-23:
Hi @tomek-l Everything is fine now and trying to upgrade to your program. At first I got an error in camera and tried to check the camera with this command ls /dev/video0. Anything have changed and got an error again, then try to install the video 4 linux utils with this command (sudo apt-get install v4l-utils) and then check video list (v4l2-ctl --list-devices). The output was fine. Thanks a lot for your kindly respond.

LXPWing, 2021-03-24:
Hi @Henery1 What version of torch , torchvision and python are you using? I'm having a problem with the version

Henery1, 2021-06-02:
Hi Tomek-I!!! I want to get the Tavg data. But I don't know how can i get this Tavg data in your program. Where can i find the data set in your program? Thanks a lot!!! Waiting for your kindly reply.

Henery1, 2021-06-02:
> 
> 
> Hi @Henery1 What version of torch , torchvision and python are you using? I'm having a problem with the version

I used Torch version 1.8.0 .