tomasz-lewicki/ai-thermometer #34: Import Error make_imx219_capture

rtime2awaken, 2021-05-04:
I'm stepping through main.py with pdb and getting: ImportError: cannot import name 'make_imx219_capture' 
I'm hoping there is a simple fix.  It looks like it has to to with the Pi camera.  I also have the lepton 3.5.  Any advice
would be much appreciated.

tomek-l, 2021-05-06:
Hi @rtime2awaken !

This looks like the RGB frame didn't get grabbed correctly. Could you confirm that you're able to see camera output by doing either:

```shell
gst-launch-1.0 nvarguscamerasrc sensor_id=0 ! 'video/x-raw(memory:NVMM),width=3264, height=2464, framerate=21/1, format=NV12' ! nvvidconv flip-method=2 ! 'video/x-raw,width=960, height=720' ! nvvidconv ! nvegltransform ! nveglglessink -e
```

or 

```shell
gst-launch-1.0 nvarguscamerasrc sensor_id=1 ! 'video/x-raw(memory:NVMM),width=3264, height=2464, framerate=21/1, format=NV12' ! nvvidconv flip-method=2 ! 'video/x-raw,width=960, height=720' ! nvvidconv ! nvegltransform ! nveglglessink -e
```

Let me know if you can see anything on the screen.

rtime2awaken, 2021-05-06:
> Hi @rtime2awaken !
>
> This looks like the RGB frame didn't get grabbed correctly. Could you confirm that you're able to see camera output by doing either:
>
> ```shell
> gst-launch-1.0 nvarguscamerasrc [...]

It has been fixed!  I was using the v1 Pi camera that was causing the problem.  Thank you for getting back with me.  This is a great project.

closed 2021-05-06 by rtime2awaken