tomasz-lewicki/ai-thermometer #35: Using an USB camera

jjferreira, 2021-05-05:
Hi there how can  make this code run using a usb camera, instead of the camera suggested in your article.

tomek-l, 2021-05-06:
Hi @jjferreira !

1) You're probably better off using CSI camera. Using a USB camera will need more CPU to copy data & decode the video stream from USB.
2) If you still want to use a USB camera here's how you'd do it:

Substitute [this line](https://github.com/tomek-l/jetson-cameras/blob/d8ff3729c9bd9d6eb06e9339aecbdc856878b2cf/__init__.py#L37) from
```python
cap = cv2.VideoCapture(gst_string, cv2.CAP_GSTREAMER)
```

to
```python
cap = cv2.VideoCapture('/dev/video3')
```

This assumes that your USB camera is on `/dev/video3`. If you don't know which device node the USB cam is associated with, do `dmesg -w` and observe the output while plugging the USB camera.

Hope this helps!

jjferreira, 2021-05-07:
Hi @tomek-l , thank you for this response, but I'm a bit confused.  This file u asked me to make a change its on a different project and repository.  I downloaded the **ai-thermometer** repository and the changes are on the **jetson-cameras** repository, do you mind clarifying.

Thank you

tomek-l, 2021-05-14:
Hi @jjferreira ,

What webcam are you using?
Did you have any success so far in getting the image in other applications?

jjferreira, 2021-05-14:
I tried my intel realsense d435i, and a standard logitech Usb camera.  Same
behaviour.  But if i run a simple python script i can see both cameras
working as expected.

Tks

jjferreira, 2021-06-28:
The issued I was having was to do with the lens of the thermal became loose.