tomasz-lewicki/ai-thermometer #40: Hi there, how do I change both images like 180 degress?

jjferreira, 2021-06-28:
Hi there, how do I change both images like 180 degress?

__Originally posted by @jjferreira in https://github.com/tomek-l/ai-thermometer/issues/39__

jjferreira, 2021-06-30:
@tomek-l hello, did you had the chance to figure out this issue?  Thank you

tomek-l, 2021-06-30:
Hi @jjferreira !

1. To rotate the RGB image by 180 degrees, you can set `flip_method=0` in [this line](https://github.com/tomek-l/JetCameras/blob/d8ff3729c9bd9d6eb06e9339aecbdc856878b2cf/__init__.py#L34)
2. To rotate the IR image by 180 degrees, you can do `self._raw = np.flipud(raw)` in [this line](https://github.com/tomek-l/ai-thermometer/blob/90ede30626143b9b13daf9920106829d614b2058/ir/purethermal.py#L174)\

Hope this helps.

jjferreira, 2021-07-01:
@tomek-l thank you for your help.  The rgb camera is via usb, so that string got no effect.

jjferreira, 2021-07-01:
@tomek-l also the change, self._raw = np.flipud(raw) does to seem to work, or st least is not represented on the UI, pls advise