tomasz-lewicki/ai-thermometer #42: Camera Issue

BasilHarun, 2021-08-16:
[image]

can you help me with this erroor

tomek-l, 2021-08-16:
Hi there!

This should help :slightly_smiling_face: 
```shell
echo 'SUBSYSTEM=="usb",  ENV{DEVTYPE}=="usb_device", GROUP="plugdev", MODE="0664"' | sudo tee /etc/udev/rules.d/10-libuvc.rules 
sudo udevadm trigger
```

You can read more about this in the [FAQ](https://github.com/tomek-l/ai-thermometer/blob/master/README.md#faqcommon-issues)

BasilHarun, 2021-08-16:
Hi, 

Thanks for replying, I’ve tried it out but see everytime it gives different error 
resize error is due to it doesn't detect pi camera
and waiting for ir frames is due to empty pipline of ir thread

[image]

The only thing I think I’m using Nano 2Gb could it be because of the memory.

BasilHarun, 2021-08-16:
[image]
[image]

BasilHarun, 2021-08-16:
[image]

tomek-l, 2021-08-16:
Yeah, so those are two separate issues:
1) with Pi Camera 
2) with USB permissions for Lepton 3.

To try addressing 1)

I'd start by restarting argus:
```shell
sudo systemctl restart nvargus-daemon.service
```
After restart, try this:
```shell
gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),  width=(int)1920, height=(int)1080, format=(string)NV12,  framerate=(fraction)30/1' ! nvoverlaysink
```
Can you see video on the screen?

Regarding 2): are you still seeing `uvc_open error -3`?

BasilHarun, 2021-08-16:
Hi, 

I did all the steps accordingly, and can see the video. 

Then I ran Python3 Main.py, 

on side note (I had a low memory warning on right hand corner.)

this is the result. 

[image]

[image]

BasilHarun, 2021-08-17:
After fixing RGB Error, it gives me the IR error. 

[image]

BasilHarun, 2021-08-17:
Also I’m Using Jetson Nano 2GB not 4GB and when I run main.py it always gives me this error on side. 

[image]

BasilHarun, 2021-08-17:
When I run the IR camera individually, I get this empty error. 

[image]

closed 2021-08-18 by BasilHarun