tomasz-lewicki/ai-thermometer #43: Low Memory Warning

BasilHarun, 2021-08-18:
Hi Guys, I’m getting this Error after executing the Main.py. 

[image]

I believe it’s a low Memory Error as I’m using Jetson Nano 2GB. 

[image]

Can anyone advise, how I can Resolve this with out changing to 4GB Nano Kit. 

Kind Regards,

tomek-l, 2021-08-18:
Hi @BasilHarun
You can increase swap size. [Here](https://www.jetsonhacks.com/2019/11/28/jetson-nano-even-more-swap/) you can read on how to do it. A lot of memory is allocated by pytorch/cuDNN, but never used in the main loop. In my case over 1GB was moved to swap on Nano 2GB.

BasilHarun, 2021-08-18:
Hi @tomek-l 

thanks for replying. I’ve already created a 5gb swap. 

[image]

but Error is persistent.

tomek-l, 2021-08-18:
As long as you have some memory left, the program shouldn't crash. The error you're seeing is generated by nvarguscamerasrc.
1. Try `sudo systemctl restart nvargus-daemon.service`, that could help.
2. What does the output of `dmesg | grep imx219` command say?