tomasz-lewicki/ai-thermometer #45: USB camera issue

immkapoor, 2021-11-04:
I am using a simple logitech usb camera with jetson xavier but I am getting a uvc_find_error error but the camera is getting detected and activated. Everything is working well with simple codes.

immkapoor, 2021-11-07:
Can you please provide me the dependencies for this repo? Which versions of opencv and pytorch are you using?

tomek-l, 2021-11-09:
Hi!

The uvc_error is likely due to the purethermal module not getting detected
(which is also a UVC device).

As far as the versioning goes:
- OpenCV> 4.4
- pytorch > 1.9

The latest releases of both should work.

immkapoor, 2021-11-09:
@tomek-l Thank you for responding. I am using a FLIR E75 camera along with legion usb camera. Both are getting detected by other codes but i am getting a device error while running this code. I am using opencv v4.4.0. I am getting warnings from gstreamer. Can you suggest anything? It'll be helpful. 
[image]
r

tomek-l, 2021-11-09:
Hi @immkapoor,

Here you would change:
https://github.com/tomek-l/ai-thermometer/blob/6be49fe3880a3381ed5b66a7fcbf8b74992ab476/rgb/rgb_thread.py#L24

To something like:
```python
self._stream = cv2.VideoCapture("/dev/video1")  
```
Where `/dev/video1` will be device handle to your USB webcam.

tomek-l, 2021-11-09:
I implemented this project with Raspberry Pi CM V2.1 which is a CSI _(Camera Serial Interface)_ camera, which has different API than your USB camera - that's why you're getting errors.

Keep in mind that when you get these two to work without errors, you will still have to make sure that the images from the two cameras "align". It's fairly simple for FLIR Lepton 3.5 and Rpi CM V2.1, because they have very similar field-of-view.

immkapoor, 2021-11-09:
@tomek-l I made the suggested changes. But I am still getting the same error. The usb camera is getting enabled but I am not able to see the camera's capture screen.  Do you think it's because of warnings raised by gstreamer
[image]
?

tomek-l, 2021-11-09:
That's odd. Did you remove the call to `make_imx219_capture()`?
That's the line of code that initializes gstreamer pipeline.

immkapoor, 2021-11-09:
@tomek-l No, i haven't made any changes in code. Just added legion camera in init.py and flir camera in rgb_thread

immkapoor, 2021-11-14:
@tomek-l Hey, I was able to solve the Gstreamer warning by replacing cv2.VIdeoCapture("/dev/video*") in cv2.VIdeoCapture("/dev/video", cv2.CAP_V4L2) and if I do make_imx_capture() then I get uvc device error and program terminates. If I replace that with another simple video capture then also I get ucv find device error but nothing comes on screen. Do you have any idea?

immkapoor, 2021-11-16:
@tomek-l I am not able to find the find device error in init code of libuvc wrapper. Can you please tell me where its definition is coming from?

tomek-l, 2021-11-16:
Hi @immkapoor 

These two are two separate methods of creating a camera device:
- `cv2.VIdeoCapture()` 
- `make_imx_capture()`

You have to choose just one. Since you are working with a USB camera, I recommend doing `cv2.VIdeoCapture("/dev/video0", cv2.CAP_V4L2)` and deleting the `make_imx_capture()` call.

tomek-l, 2021-11-16:
As far as the error codes go, you can look them up in [libuvc](https://github.com/groupgets/libuvc) source code.

Specifically, in [libuvc.h](https://github.com/groupgets/libuvc/blob/5e866910de3f172bd5adab72baec40a066bbbe3a/include/libuvc/libuvc.h#L23) you have definitions of what each error number means.