tomasz-lewicki/ai-thermometer #47: How to get started

geannealva, 2021-12-15:
Hi @tomek-l ! I hope you could help me.

I am a graduating college student. I have a thesis about face mask detection and body temperature detection.. I am currently working on my model for the mask detection, however I really do not know how to get started with the temperature detection. I also have the same idea which is to detect the inner eye canthus for temperature measurement, (using haar as well?). I am also planning to buy the same materials. But I really don't know how the code works or how I will calibrate the cameras. I am still a beginner in machine learning with python because I studied on my own. I hope you could help me, I really want to learn. This is the only project I found helpful. Thank you so much! 😅

rjv-vila, 2021-12-16:
@geannealva I think I can help you since we have the same project as I posted in Issue#46. You can send me an email.
Josebustamante2019@gmail.com

tomek-l, 2021-12-16:
Hi @geannealva !

As much as I'd love to help, I probably don't have the throughput to do that :slightly_smiling_face: 
If you have specific questions, feel free to ask them here.

tomek-l, 2021-12-16:
Referring to the question from the topic: "How to get started".

The starting point I'd recommend is to:

1. Buy the IR camera
2. Try running this script [pt2_example.py](https://github.com/tomek-l/ai-thermometer/blob/6be49fe3880a3381ed5b66a7fcbf8b74992ab476/ir/pt2_example.py#L106). At [Line 106](https://github.com/tomek-l/ai-thermometer/blob/6be49fe3880a3381ed5b66a7fcbf8b74992ab476/ir/pt2_example.py#L106) `temp_map` will be a numpy array in Celsius. Try measuring the temperature of objects in front of the IR camera.
3. Once you have IR and RGB cameras running, go to [Getting started with cameras](https://github.com/tomek-l/ai-thermometer/blob/master/docs/notebooks/Getting%20Started%20with%20Cameras.ipynb) notebook and try combining IR and RGB footage.

geannealva, 2021-12-16:
@rjv-vila thank you so much!

geannealva, 2021-12-16:
@tomek-l thank you so much!

rjv-vila, 2022-01-20:
@geannealva My email above was hacked. If you're from PH , you can contact me in anbuflame16@gmail.com

yagmurarsln0509, 2022-08-01:
Hello, need some advice please.. I run this script pt2_example.py. I measure the temperature of objects in front of the IR camera. But  I am getting syntax error when executing Getting started with cameras.ipynb example. 
!gst-launch-1.0 v4l2src device=/dev/video1 num-buffers=1 ! video/x-raw,format=GRAY16_LE ! videoconvert ! pngenc ! filesink location=/tmp/ir.png
ı run code in the shell. So I  remove the preceding '!' from the command.
gst-launch-1.0 v4l2src device=/dev/video1 num-buffers=1 ! video/x-raw,format=GRAY16_LE ! videoconvert ! pngenc ! filesink location=/tmp/ir.png
out : SyntaxError: invalid syntax
Please . Can you help me ?